espressif/crosstool-NG #6: expat-2.2.5 cannot be downloaded

XaviDCR92, 2020-11-06:
### Issue submission guidelines
See [detaled guidelines](http://crosstool-ng.github.io/support/). In short:
- Report crosstool-NG version
`esp-2020r3`
- Report host OS and its version
`x86_64 Kubuntu 20.04 LTS`
- Attach `.config` (unless the issue is not related to any configuration)
- Attach `build.log` (unless crosstool-NG fails before the build starts)
[build.log](https://github.com/espressif/crosstool-NG/files/5501410/build.log)

I have followed the [instructions from esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup-scratch.html) and the script is unable to download `expat-2.2.5` because of `Unable to establish SSL connection`.

XaviDCR92, 2020-11-12:
I have tried again today and it looks like a temporary issue, since now `expath-2.2.5` has downloaded successfully.

closed 2020-11-12 by XaviDCR92

lypanov, 2021-04-02:
This is now broken and reproducibly so.

Quick hack is replacing the CT_EXPAT_ARCHIVE_FILENAME definition in .config with:
`CT_EXPAT_ARCHIVE_FILENAME="@{pkg_name}-@{version}-RENAMED-VULNERABLE-PLEASE-USE-2.3.0-INSTEAD"`

This is of course really not something that users should need to do.

Only doing this because an Raspberry Pi compatible release is still not available.

antmak, 2021-04-03:
Hello, @lypanov 

> Only doing this because a Raspberry Pi compatible release is still not available.

Which version RPi do you use? `armel` options from  https://github.com/espressif/crosstool-NG/releases work fine with RPi 3 and 4. So, when you use IDF and its `idf_tools.py`, it would download a right platform version on your host.

antmak, 2021-04-03:
> Quick hack is replacing the CT_EXPAT_ARCHIVE_FILENAME definition in .config with:
> `CT_EXPAT_ARCHIVE_FILENAME="@{pkg_name}-@{version}-RENAMED-VULNERABLE-PLEASE-USE-2.3.0-INSTEAD"`

thanks for the info!

lypanov, 2021-04-03:
> Hello, @lypanov
>
>
> Which version RPi do you use? `armel` options from https://github.com/espressif/crosstool-NG/releases work fine with RPi 3 and 4. So, when you use IDF and its `idf_tools.py`, it [...]

Yeah good point, my issue is rather niche I guess. I'm running 64 bit RPi via Ubuntu rather than running the out of the box 32bit OS.