espressif/crosstool-NG #8: esp-2020r3 code for ESP8266 RTOS SDK is not available

trombik, 2021-01-13:
### Issue submission guidelines

- crosstool-NG version N/A
- host OS and its version N/A

with https://github.com/espressif/ESP8266_RTOS_SDK/commit/fd54106a97a8d2abdc891ff64d4f5c4c4eba4235, the latest ESP8266 RTOS SDK requires esp-2020r3 version (gcc 8.x) of toolchain. however, the code is not available. the last commit in [esp8266-1.22.x](https://github.com/espressif/crosstool-NG/commits/esp8266-1.22.x) branch was made in 2019. this is not only inconvenience (i cannot build the toolchain), but also license violation, IIRC.

slightly off-topic: `esp-idf` is probably one of the best OSS SDK by vendors. the development team is very responsible, listen to customers and users, quick to fix issues in general, and writes great documentations. however, `ESP8266 RTOS SDK` is just like many products by Chinese vendors. [Issues are ignored](https://github.com/espressif/ESP8266_RTOS_SDK/issues), documentation is sparse, sub-modules are left behind (`esp-mqtt` is a good example). I do understand espressif  does not want to invest resources on legacy products, but hope it maintains the SDK with a little bit more love.

antmak, 2021-01-13:
Hi @trombik ! Thanks for your notices.

  Regarding the first part, the sources of the latest ESP8266 toolchain was published in [esp-develop](https://github.com/espressif/crosstool-NG/commits/esp-develop) branch ( the mentioned release was a `esp-2020r3-49-gd5524c1` or d5524c19620c24b5ae700c2e33817610351df37c commit). Probably, it should be better if the changes could be placed in `esp-1.23.x` branch too (where we collected all chips together). `esp8266-1.22.x` branch is based on the old crosstool-ng.

antmak, 2021-01-13:
I've updated [esp-1.23.x](https://github.com/espressif/crosstool-NG/commits/esp-1.23.x) branch to be more consistent in `esp-2020r3-49-gd5524c1` release's tags, https://github.com/espressif/crosstool-NG/commit/a44a340398f3025fa9611444fe28ca2f6ede6fba

trombik, 2021-01-14:
@antmak thanks!

closed 2021-01-14 by trombik