espressif/crosstool-NG #11: Feature request: c++20 support

blaizard, 2021-03-14:
Is there any plan to add partial support to c++20?
I am especially interested in support from coroutines which comes with gcc 10.
If not what does it take to upgrade gcc?

antmak, 2021-03-22:
Hi @blaizard ,
This is a good and an expected thing, we'll add GCC 10 support in some follow-up releases of toolchain and IDF. I've tried to build a toolchain with GCC 10, and found some issues with the current IDF code.

You can find some build details in [build docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/linux-setup-scratch.html), https://github.com/antmak/crosstool-ng/commits/esp-1.24.x-gcc10 , https://github.com/antmak/gcc/commits/esp_based_on_10_2_0 , builds for [win](https://dl.espressif.com/dl/toolchains/preview/gcc10/xtensa-esp32-elf-gcc10_2_0-crosstool-ng-1.24.0-302-g32e30a1-win64.zip), [lin](https://dl.espressif.com/dl/toolchains/preview/gcc10/xtensa-esp32-elf-gcc10_2_0-crosstool-ng-1.24.0-302-g32e30a1-linux-amd64.tar.gz), [macos](https://dl.espressif.com/dl/toolchains/preview/gcc10/xtensa-esp32-elf-gcc10_2_0-crosstool-ng-1.24.0-302-g32e30a1-macos.tar.gz) These are unofficial resources, no any warranty

b1ackviking, 2021-03-26:
> we'll add GCC 10 support in some follow-up releases of toolchain

Thanks, I'm looking forward to it.

p5-vbnekit, 2021-06-08:
It would be great to use c++ standard `>= c++20` (coroutines, concepts, etc). Are there any chances that the toolchain will get updated? =)

b1ackviking, 2021-06-09:
@p5-vbnekit Coroutines, concepts, constexpr algorithms are part of C++20.

b1ackviking, 2021-06-16:
I am not sure if I should put it here or open a separate issue, but I am facing the problem of the standard library lacking some older C++ features.

As far as I know, the current GCC version (8.4.0) should support C++11, C++14, and almost everything of C++17 (except the `<memory_resource>` header). But last week I found out that the standard library misses some algorithms, like `std::for_each_n` (C++17). And `std::initializer_list` does not have `rbegin`/`rend` (C++14).

@antmak Maybe you can provide any comments on that, please?

Emill, 2021-06-21:
> This is a good and an expected thing, we'll add GCC 10 support in some follow-up releases of toolchain and IDF. I've tried to build a toolchain with GCC 10, and found some issues with the current IDF code.

What are the issues found?