espressif/crosstool-NG #14: Feature request: include archive sizes in releases sha256 file

me-no-dev, 2021-06-28:
Arduino requires archive sizes in it's [configuration](https://github.com/espressif/arduino-esp32/blob/master/package/package_esp32_index.template.json#L84). Currently I have to download all archives just to grab the correct file sizes. I hope it's possible to include that in the sha256 file in the releases, so I do not have to download everything in the future :)

igrr, 2021-06-28:
@me-no-dev until this is implemented, I might suggest to do a request to https://api.github.com/repos/espressif/crosstool-ng/releases/tags/esp-2021r1 — each element of the "assets" array contains a "size" field.

me-no-dev, 2021-06-29:
This will work for now :) thanks!

antmak, 2021-11-09:
The sizes were added to *-checksum.sha256 file since 2021r1 release, thanks @me-no-dev for noticing that

closed 2021-11-09 by antmak