espressif/crosstool-NG #17: Retrieving isl-0.19: download failed

NoUmlautsAllowed, 2021-11-02:
I followed the [Setup Linux Toolchain from scratch](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup-scratch.html) tutorial and noticed the following issue.

It seems like the mirror `http://isl.gforge.inria.fr` is down. Therefore, `./ct-ng build` fails with this error (also see attached log below):
```
[EXTRA]    Retrieving 'isl-0.19'
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.19.tar.xz'
[DEBUG]    ==> Executing:  'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/esp/crosstool-NG/.build/tarballs/isl-0.19.tar.xz.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.19.tar.xz'
[ALL  ]    --2021-11-02 11:39:06--  http://isl.gforge.inria.fr/isl-0.19.tar.xz
[ALL  ]    Resolving isl.gforge.inria.fr (isl.gforge.inria.fr)... 128.93.193.15
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2021-11-02 11:39:17--  (try: 2)  http://isl.gforge.inria.fr/isl-0.19.tar.xz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2021-11-02 11:39:29--  (try: 3)  http://isl.gforge.inria.fr/isl-0.19.tar.xz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Giving up.
[ALL  ]    
[DEBUG]    ==> Return status 4
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.19.tar.xz"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.19.tar.bz2'
[DEBUG]    ==> Executing:  'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/esp/crosstool-NG/.build/tarballs/isl-0.19.tar.bz2.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.19.tar.bz2'
[ALL  ]    --2021-11-02 11:39:39--  http://isl.gforge.inria.fr/isl-0.19.tar.bz2
[ALL  ]    Resolving isl.gforge.inria.fr (isl.gforge.inria.fr)... 128.93.193.15
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2021-11-02 11:39:50--  (try: 2)  http://isl.gforge.inria.fr/isl-0.19.tar.bz2
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2021-11-02 11:40:02--  (try: 3)  http://isl.gforge.inria.fr/isl-0.19.tar.bz2
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Giving up.
[ALL  ]    
[DEBUG]    ==> Return status 4
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.19.tar.bz2"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.19.tar.gz'
[DEBUG]    ==> Executing:  'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/esp/crosstool-NG/.build/tarballs/isl-0.19.tar.gz.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.19.tar.gz'
[ALL  ]    --2021-11-02 11:40:12--  http://isl.gforge.inria.fr/isl-0.19.tar.gz
[ALL  ]    Resolving isl.gforge.inria.fr (isl.gforge.inria.fr)... 128.93.193.15
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2021-11-02 11:40:23--  (try: 2)  http://isl.gforge.inria.fr/isl-0.19.tar.gz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2021-11-02 11:40:35--  (try: 3)  http://isl.gforge.inria.fr/isl-0.19.tar.gz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Giving up.
[ALL  ]    
[DEBUG]    ==> Return status 4
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.19.tar.gz"
[ERROR]    isl: download failed
```

I resolved this issue locally by replacing the mirror `http://isl.gforge.inria.fr` with `https://libisl.sourceforge.io` in `config/versions/isl.in` like this:

```shell
$ diff isl.in.orig isl.in
528c528
<     default "http://isl.gforge.inria.fr"
---
>     default "https://libisl.sourceforge.io"
```

Unfortunately I am not that familiar with crosstool-NG, so I do not know where to change the mirror of libisl before generating the configuration files.

Thanks in advance.

Best regards
NoU

-----------------------------------

Further information:
 - crosstool-NG version: 
   ```
    $ ./ct-ng --version
    GNU Make 4.2.1
    Built for x86_64-pc-linux-gnu
    Copyright (C) 1988-2016 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
   ```

 - Host OS and version:
   ```
    $ uname -a
    Linux 808b49755bd8 5.11.0-38-generic #42~20.04.1-Ubuntu SMP Tue Sep 28 20:41:07 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
   ```
 - Attached [build.log](https://github.com/espressif/crosstool-NG/files/7459948/build.log)

NoUmlautsAllowed, 2021-11-03:
Ok, I found the configuration of the mirror of the isl package. The changes should fix the error described above.

dev-null2019, 2021-11-05:
Same problem here - thanks for the fix!

dev-null2019, 2021-11-05:
Same problem here!

`[EXTRA]    Retrieving 'isl-0.19'
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.19.tar.xz'
[DEBUG]    ==> Executing:  'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/stefan/esp/crosstool-NG/.build/tarballs/isl-0.19.tar.xz.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.19.tar.xz'
[ALL  ]    --2021-11-05 19:56:52--  http://isl.gforge.inria.fr/isl-0.19.tar.xz
[ALL  ]    Resolving isl.gforge.inria.fr (isl.gforge.inria.fr)... 128.93.193.15
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]
[ALL  ]    --2021-11-05 19:57:03--  (try: 2)  http://isl.gforge.inria.fr/isl-0.19.tar.xz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]
[ALL  ]    --2021-11-05 19:57:15--  (try: 3)  http://isl.gforge.inria.fr/isl-0.19.tar.xz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Giving up.
[ALL  ]
[DEBUG]    ==> Return status 4
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.19.tar.xz"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.19.tar.bz2'
[DEBUG]    ==> Executing:  'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/stefan/esp/crosstool-NG/.build/tarballs/isl-0.19.tar.bz2.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.19.tar.bz2'
[ALL  ]    --2021-11-05 19:57:26--  http://isl.gforge.inria.fr/isl-0.19.tar.bz2
[ALL  ]    Resolving isl.gforge.inria.fr (isl.gforge.inria.fr)... 128.93.193.15
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]
[ALL  ]    --2021-11-05 19:57:37--  (try: 2)  http://isl.gforge.inria.fr/isl-0.19.tar.bz2
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]
[ALL  ]    --2021-11-05 19:57:49--  (try: 3)  http://isl.gforge.inria.fr/isl-0.19.tar.bz2
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Giving up.
[ALL  ]
[DEBUG]    ==> Return status 4
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.19.tar.bz2"
[DEBUG]    Trying 'http://isl.gforge.inria.fr/isl-0.19.tar.gz'
[DEBUG]    ==> Executing:  'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/stefan/esp/crosstool-NG/.build/tarballs/isl-0.19.tar.gz.tmp-dl' 'http://isl.gforge.inria.fr/isl-0.19.tar.gz'
[ALL  ]    --2021-11-05 19:58:00--  http://isl.gforge.inria.fr/isl-0.19.tar.gz
[ALL  ]    Resolving isl.gforge.inria.fr (isl.gforge.inria.fr)... 128.93.193.15
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]
[ALL  ]    --2021-11-05 19:58:11--  (try: 2)  http://isl.gforge.inria.fr/isl-0.19.tar.gz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]
[ALL  ]    --2021-11-05 19:58:23--  (try: 3)  http://isl.gforge.inria.fr/isl-0.19.tar.gz
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... failed: Connection timed out.
[ALL  ]    Giving up.
`
`$ uname -a
Linux j710 3.16.81 #39 Wed Oct 27 20:43:32 CEST 2021 armv4l GNU/Linux`

`./ct-ng --version
GNU Make 4.1
Gebaut für arm-unknown-linux-gnueabi
Copyright (C) 1988-2014 Free Software Foundation, Inc.
Lizenz GPLv3+: GNU GPL Version 3 oder später <http://gnu.org/licenses/gpl.html>
Dies ist freie Software: Sie können sie nach Belieben ändern und weiter verteilen.
Soweit es die Gesetze erlauben gibt es KEINE GARANTIE.`

antmak, 2021-11-09:
Fixed in the current [esp-1.24.x](https://github.com/espressif/crosstool-NG/commits/esp-1.24.x) branch, abb32e6db4cb86ee344b710693441208b1cf09bf

closed 2021-11-09 by antmak