espressif/crosstool-NG #19: configure: error: no usable python found at python2

dev-null2019, 2021-11-10:
Hi!
Tried to do a Canadian build on Devuan Beowulf with this configuration:

```
[EXTRA]    Building a toolchain for:
[EXTRA]      build  = x86_64-pc-linux-gnu
[EXTRA]      host   = arm-linux-gnueabi
[EXTRA]      target = xtensa-esp32-elf
```
```
$ ./ct-ng --version
Copyright (C) 1988-2016 Free Software Foundation, Inc.  
Lizenz GPLv3+: GNU GPL Version 3 oder später <http://gnu.org/licenses/gpl.html>  
Dies ist freie Software: Sie können sie nach Belieben ändern und weiter verteilen.  
Soweit es die Gesetze erlauben gibt es KEINE GARANTIE.  
GNU Make 4.2.1  
Gebaut für x86_64-pc-linux-gnu  
```

```
$ uname -a
Linux spndevuan3work 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 (2021-09-29) x86_64 GNU/Linux
```

Everything went fine until it tries to find python2.
```
[ERROR]    configure: error: no usable python found at python2  
[ERROR]    make[2]: *** [Makefile:9533: configure-gdb] Error 1  
[ERROR]    make[2]: *** Waiting for unfinished jobs....  
[ALL  ]    mv -f .deps/ctf-create.Tpo .deps/ctf-create.Plo  
[ALL  ]    /bin/bash ./libtool  --tag=CC   --mode=link arm-host_unknown-linux-gnueabi-gcc -std=gnu99 -Wall -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -pedantic -Wno-long-long  -I/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/src/gdb/libctf/../zlib -O2 -g -pipe -I/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/buildtools/complibs-host/include -version-info 0:0:0  -Wl,--version-script='/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/src/gdb/libctf/libctf.ver' -L/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/buildtools/complibs-host/lib -o libctf.la  ctf-archive.lo ctf-dump.lo ctf-create.lo ctf-decl.lo ctf-error.lo ctf-hash.lo ctf-labels.lo ctf-link.lo ctf-lookup.lo ctf-open.lo ctf-string.lo ctf-subr.lo ctf-types.lo ctf-util.lo  ctf-open-bfd.lo -L./../zlib -lz -ldl   
[ALL  ]    /bin/bash ./libtool  --tag=CC   --mode=link arm-host_unknown-linux-gnueabi-gcc -std=gnu99 -Wall -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -pedantic -Wno-long-long  -I/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/src/gdb/libctf/../zlib -O2 -g -pipe -I/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/buildtools/complibs-host/include -version-info 0:0:0  -Wl,--version-script='/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/src/gdb/libctf/libctf.ver' -L/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/buildtools/complibs-host/lib -o libctf-nobfd.la  ctf-archive.lo ctf-dump.lo ctf-create.lo ctf-decl.lo ctf-error.lo ctf-hash.lo ctf-labels.lo ctf-link.lo ctf-lookup.lo ctf-open.lo ctf-string.lo ctf-subr.lo ctf-types.lo ctf-util.lo  -L./../zlib -lz -ldl   
[ALL  ]    libtool: link: warning: -version-info/-version-number' is ignored for convenience libraries  
[ALL  ]    libtool: link: warning: -version-info/-version-number' is ignored for convenience libraries  
[ALL  ]    libtool: link: arm-host_unknown-linux-gnueabi-ar rc .libs/libctf.a  ctf-archive.o ctf-dump.o ctf-create.o ctf-decl.o ctf-error.o ctf-hash.o ctf-labels.o ctf-link.o ctf-lookup.o ctf-open.o ctf-string.o ctf-subr.o ctf-types.o ctf-util.o ctf-open-bfd.o  
[ALL  ]    libtool: link: arm-host_unknown-linux-gnueabi-ar rc .libs/libctf-nobfd.a  ctf-archive.o ctf-dump.o ctf-create.o ctf-decl.o ctf-error.o ctf-hash.o ctf-labels.o ctf-link.o ctf-lookup.o ctf-open.o ctf-string.o ctf-subr.o ctf-types.o ctf-util.o  
[ALL  ]    libtool: link: arm-host_unknown-linux-gnueabi-ranlib .libs/libctf.a  
[ALL  ]    libtool: link: arm-host_unknown-linux-gnueabi-ranlib .libs/libctf-nobfd.a  
[ALL  ]    libtool: link: ( cd ".libs" && rm -f "libctf.la" && ln -s "../libctf.la" "libctf.la" )  
[ALL  ]    libtool: link: ( cd ".libs" && rm -f "libctf-nobfd.la" && ln -s "../libctf-nobfd.la" "libctf-nobfd.la" )  
[ALL  ]    make[4]: Leaving directory '/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/build/build-gdb-cross/libctf'  
[ALL  ]    make[3]: Leaving directory '/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/build/build-gdb-cross/libctf'  
[ALL  ]    make[2]: Leaving directory '/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/build/build-gdb-cross'  
[ERROR]    make[1]: *** [Makefile:851: all] Error 2  
[ALL  ]    make[1]: Leaving directory '/home/stefan/esp/crosstool-NG/.build/HOST-arm-linux-gnueabi/xtensa-esp32-elf/build/build-gdb-cross'  
[ERROR]    
[ERROR]  >>  
[ERROR]  >>  Build failed in step 'Installing cross-gdb'  
[ERROR]  >>        called in step '(top-level)'  
[ERROR]  >>  
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@376]  
[ERROR]  >>        called from: do_gdb_backend[scripts/build/debug/300-gdb.sh@281]  
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@59]  
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]  
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@695]  
[ERROR]  >>  
[ERROR]  >>  For more info on this error, look at the file: 'build.log'  
[ERROR]  >>  There is a list of known issues, some with workarounds, in:  
[ERROR]  >>      https://crosstool-ng.github.io/docs/known-issues/  
[ERROR]  >>  
[ERROR]  >> NOTE: Your configuration includes features marked EXPERIMENTAL.  
[ERROR]  >> Before submitting a bug report, try to reproduce it without enabling  
[ERROR]  >> any experimental features. Otherwise, you'll need to debug it  
[ERROR]  >> and present an explanation why it is a bug in crosstool-NG - or  
[ERROR]  >> preferably, a fix.  
[ERROR]  >>  
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:  
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/  
[ERROR]  >>  
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.  
[ERROR]  >>  Read the bug reporting guidelines here:  
[ERROR]  >>      http://crosstool-ng.github.io/support/  
[ERROR]    
[ERROR]  (elapsed: 77:16.08)```

```

```
$ /usr/bin/python2 --version
Python 2.7.16

$ /usr/bin/python3 --version
Python 3.7.3

```

Python 2 and 3 are installed and also package python-dev is installed - Python works well on this system.

[config.txt](https://github.com/espressif/crosstool-NG/files/7510591/config.txt)

antmak, 2021-11-19:
Can you make sure the package `python-dev` is installed?

antmak, 2021-11-19:
It's a canadian build... Please try `echo "# CT_GDB_CROSS_PYTHON is not set" >> .config` and `./ct-ng oldconfig && ./ct-ng build`. https://github.com/espressif/crosstool-NG/blob/abb32e6db4cb86ee344b710693441208b1cf09bf/.gitlab-ci.yml#L83

dev-null2019, 2021-11-29:
Hello!
This doesnt work.

[INFO ]  Extracting and patching toolchain components
[ERROR]    Overlay /home/stefan/esp/crosstool-NG/overlays/xtensa_esp32 not found
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@487]
[ERROR]  >>        called from: CT_DoExtractPatch[scripts/functions@2253]
[ERROR]  >>        called from: CT_PackageRun[scripts/functions@2085]
[ERROR]  >>        called from: CT_ExtractPatch[scripts/functions@2421]
[ERROR]  >>        called from: do_zlib_extract[scripts/build/companion_libs/050-zlib.sh@21]
[ERROR]  >>        called from: do_companion_libs_extract[scripts/build/companion_libs.sh@22]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@664]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      https://crosstool-ng.github.io/docs/known-issues/
[ERROR]  >>
[ERROR]  >> NOTE: Your configuration includes features marked EXPERIMENTAL.
[ERROR]  >> Before submitting a bug report, try to reproduce it without enabling
[ERROR]  >> any experimental features. Otherwise, you'll need to debug it
[ERROR]  >> and present an explanation why it is a bug in crosstool-NG - or
[ERROR]  >> preferably, a fix.
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]
[ERROR]  (elapsed: 7:37.22)
[07:38] / make: *** [ct-ng:261: build] Fehler 1

antmak, 2021-11-30:
@dev-null2019 

>  [ERROR] Overlay /home/stefan/esp/crosstool-NG/overlays/xtensa_esp32 not found

Please run `git submodule update --init` in working directory

dev-null2019, 2021-11-30:
Thanks you - I have successfully built it!

closed 2021-11-30 by dev-null2019