espressif/crosstool-NG #20: PlatformIO package not up to date, please update to patch2 version

FaBjE, 2022-01-18:
The latest PlatformIO package https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32/versions does not include the patch1 and patch2 versions. 

As mentioned here: https://github.com/platformio/platform-espressif32/issues/705

If you try to use the (latest) 8.4.0+2021r2 package on windows, GDB (and possibly other tools) will not work because of the missing dll's.

Please generate a PlatformIO package that contains the latest patches.

igrr, 2022-01-18:
cc @me-no-dev

Jason2866, 2022-01-18:
Please upload all actual toolchains/tools from https://github.com/espressif/esp-idf/blob/release/v4.4/tools/tools.json to platformio registry.
Thx!

FaBjE, 2022-01-21:
https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32/versions now contains 8.4.0+2021r2-patch2.
I verified it to work out of the box on Windows. (and still on Linux too)

Thank you for updating.

Closing issue.

closed 2022-01-21 by FaBjE

X-Ryl669, 2022-08-23:
Can you upload the latest toolchain 11.2.0 to the registry too ? Thanks!