espressif/crosstool-NG #21: Feature Request: Toolchain-xtensa-esp32s2 for linux_aarch64

maedball, 2022-01-25:
Can you please provide linux_aarch64 packages to the platformio platform? We need this for esphome on raspberry pis to compile firmwares.

please see the platformio feature request for reference.

https://github.com/platformio/platform-espressif32/issues/699#issuecomment-1019545134

Thank you!

Gisi0, 2022-02-06:
the same for me

deathsfaction, 2022-02-11:
Same here - so many problems now with ESPHome, etc.

manyos-alex, 2022-03-30:
any feedback on this?

Jason2866, 2022-04-14:
I have compiled under Ubuntu aarch64. Tested and verified working
https://github.com/Jason2866/crosstool-NG/releases

Taylorp913, 2022-04-27:
@Jason2866, thank you for your efforts on this for aarch64. I am waiting for this so I can compile using esphome for esp32-c3 on my Odroid N2+. 
What is going on with getting the pull request finalised?

Jason2866, 2022-04-27:
The PR is ready for merge. Espressif is checking.
For Apple Silicon you can use this
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip
```
For Linux aarch64 this
```
platform = https://github.com/Jason2866/platform-espressif32.git#203/aarch64
```

Taylorp913, 2022-04-27:
Thank @Jason2866 ! I have been trying to figure out how to source your code for a few days but had not noticed your platform-espressif32 fork. Unfortunately it did not work in my esphome (Odroid N2+). I tried it a few times with different yaml settings.

Here are the logs if you are interested. Do you think this issue is worth looking into?
 
[logs_spare-room-bulb-1_compile.txt](https://github.com/espressif/crosstool-NG/files/8578104/logs_spare-room-bulb-1_compile.txt)
[logs_spare-room-bulb-1_compile (1).txt](https://github.com/espressif/crosstool-NG/files/8578096/logs_spare-room-bulb-1_compile.1.txt)
[logs_spare-room-bulb-1_compile (2).txt](https://github.com/espressif/crosstool-NG/files/8578098/logs_spare-room-bulb-1_compile.2.txt)
.

Jason2866, 2022-04-28:
@Taylorp913 I modified just for use with Arduino. I have updated IDF to 4.4.
Before you try again delete the hidden folder `.platformio` This is needed since old stuff here will make issues. The implementation is not a clean full blown solution and do not check for old stuff left here

maedball, 2022-06-20:
> 

How to do get esphome to use the github from jason2866 to load instead of the default one?

Jason2866, 2022-06-20:
@maedball  i think i read recently that espressif compiled and uploaded official linux aarch64 toolchains to the platformio registry. So it should work with the latest platformio espressif32 release. I cant help with esphome.

maedball, 2022-06-20:
Sorry i forgott that i have to work the chain now back from espressif to ESP Home ;-)

Looks like plattformIO still has the old version:

[image]

maedball, 2022-08-06:
Was someone able to get Jasons fix to be used in esphome?

Jason2866, 2022-08-06:
@maedball  I have updated the branch to the latest changes from Tasmota upstream fork (for Apple arm64 and all others) and applied the needed changes. Since i have not a aarch64 platform reachable at the moment, the changes are untested!

Jason2866, 2022-08-10:
@maedball  The aarch64 toolchains ARE in the Platformio registry. Use latest Platformio  espressif32 release and you can build on linux aarch64. Nothing else needs to be done.
I will delete my not anymore needed hombrew fork.
[image]

maedball, 2022-08-13:
That is very good news. Unfortunately i am running now into the next dependencies:

If i would like to use Arduino i would need (relevant for this Feature request):
platformio/toolchain-xtensa32s2
[image]

platformio/toolchain-riscv-esp (That is for the C3)  (relevant for this Feature request)
[image]

For esp-idf i am now missing:
platformio/toolchain-esp32s2ulp  (i will raise this in the dedicated GIT)
[image]

all needed in linux_aarch64

Jason2866, 2022-08-13:
You are searching with `platformio`. The toolchains in registry are provided from `espressif`!  Every toolchain is there for `aarch64` except ulp

Jason2866, 2022-08-13:
Using latest espressif32 release does install all toolchains correctly for linux aarch64.
I compiled yesterday Tasmota under Ubuntu 22.04 (aarch64). After i managed to install Codium with (manually installed platformio.vsix and cpp.vsix). Every variant for ESP32, ESP32-S2, ESP32-S3 and ESP32-C3 of Tasmota compiled fine. Flashing the firmwares on the devices worked as expected.

closed 2024-06-11 by Lapshin