espressif/crosstool-NG #22: esp-2022r1 - release coming

antmak, 2022-04-08:
We plan another esp-2022r1 release soon for our xtensa and riscv32 chips. Main changes:

- GCC 11
- Newlib 4.1.0
- macOS@aarch64 native support
- fixed problems when running on some linux@arm platforms

Development branches where published for your info

https://github.com/espressif/crosstool-NG/commits/esp-2022r1-dev
https://github.com/espressif/gcc/commits/esp-11_2_0-dev
https://github.com/espressif/newlib-esp32/commits/esp-4.1.0-dev

(_These branches will be rebased or removed after the release_)

wuyuanyi135, 2022-04-22:
@antmak Hi! Thanks for the great work. Do you have an estimation about when this update will land?

Jason2866, 2022-06-05:
> [esp-2022r1-RC1](https://github.com/espressif/crosstool-NG/releases/tag/esp-2022r1-RC1) is coming. It will appear in IDF master soon.
> 
> _Unfortunately Apple ARM64 Native archives haven't built yet in this Release Candidate, will be added in Release_

@antmak With a small change (disable ncurses build) the setup does compile without a issue on a Apple ARM64

antmak, 2022-07-07:
@Jason2866 you're right, thank you! We didn't have time to test it properly before RC1 but we will release the native builds soon. And for v4.4 as well

antmak, 2022-07-26:
[esp-2022r1](https://github.com/espressif/crosstool-NG/releases/tag/esp-2022r1) is coming soon in IDF. There are native macOS aarch64 native builds!

antmak, 2022-09-15:
Closing it because esp-2022r1 is already in IDF master and v5.0. Also we're updating the toolchain in v4.4 these days.

The developing branches will be removed. All current code is in `esp_based_` branches and apparent tags over all related projects

closed 2022-09-15 by antmak