espressif/crosstool-NG #26: For binary built for esp32 target xtensa-esp32-elf-objdump produces incorrect disassembly, while xtensa-esp32s3-elf-objdump is correct (IDFGH-7369)

michael-sayapin, 2022-05-14:


igrr, 2022-05-14:
@michael-sayapin could you please attach the ELF file which has this issue?

michael-sayapin, 2022-05-15:
> @michael-sayapin could you please attach the ELF file which has this issue?

Let me check with the team, as it is a commercial product and I'm not sure I can open the ELF file. I will try to reproduce with a minimal example.

igrr, 2022-05-15:
Another option for you to consider is sending it to me (ivan at espressif).

igrr, 2022-05-16:
Thank you for sharing the ELF file @michael-sayapin.

Upon a brief look, it seems that there are two issues here:

1. ESP32 and ESP32-S3 builds of objdump interpret the padding "0x00" byte after the unconditional jump differently. ESP32 version of objdump interprets it along with two other bytes as an "lsi" instruction. The ESP32-S3 version of objdump interprets it as a padding byte and correctly decodes the next 3 bytes as "call8". In theory, ESP32-S3 version should behave the same way ESP32 version does (i.e. incorrectly).

   We will investigate why this happens.

2. Currently, ESP-IDF linker scripts don't add special ELF sections which contain information about such padding bytes, and which can used by the disassembler to avoid issues like this. This fact has been noticed recently by @o-marshmallow and there is a patch available internally for this issue. We haven't merged the patch yet, still investigating one issue which has surfaced when we ran our test suite.

   I'm attaching the patch here for you to try. Please try applying this patch to ESP-IDF, link the application again and see if the ESP32 disassembler starts interpreting your ELF file correctly. [17268.patch.txt](https://github.com/espressif/crosstool-NG/files/8698627/17268.patch.txt)

michael-sayapin, 2022-05-28:
Finally had a moment to try the patch. Turns out, the builder we use is based on IDF 4.2, so I had to modify the patch quite a bit to apply. In the end it applied fully. 

However, the disassembly result is the same, and seems like it made it a bit worse.

PS: Both non-patched and patched versions contain correct calls if disassembled with `-r` flag (relocation entries).