espressif/crosstool-NG #27: How to switch the toolchain version on Windows esp-idf?

wuyuanyi135, 2022-06-04:
Hello, I am using esp-idf 4.4.1 and I want to test the recently released 2022r1-rc1 toolchain. Is there any guide to make the idf.py using the new toolchain instead of the one shipped with it?

igrr, 2022-06-04:
Hi @wuyuanyi135,
esp2021-r1 toolchain is not compatible with IDF v4.4. To support GCC 11, we had to make multiple significant changes, including breaking ones. They will be released as part of IDF v5.0.
v4.4 release branch will continue to use the toolchain based on GCC 8.

wuyuanyi135, 2022-06-04:
@igrr Thanks for your detailed instructions. Does the current IDF v5.0 branch already support GCC11?

igrr, 2022-06-04:
We have just merged support for GCC 11 into IDF internally. It might take a few days for the code to get synced to Github, as currently there are some failing tests in our CI pipeline, we'll have to fix them first.

wuyuanyi135, 2022-06-04:
Much appreciated!

closed 2022-06-04 by wuyuanyi135