espressif/crosstool-NG #29: `Bad CPU type` error in platformio toolchain patch5

Jason2866, 2022-09-20:
@antmak  Looks like the package.json or the wrong toolchain is uploaded for MacOs (Intel)

Platformio log file
```
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.5 
 - tool-esptoolpy @ 1.40300.220914 (4.3.0) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting WiFiScan.ino
sh: /Users/runner/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++: Bad CPU type in executable
*** Error 1[26](https://github.com/Jason2866/platform-espressif32/actions/runs/3092921718/jobs/5004727003#step:5:27)
```

Jason2866, 2022-09-20:
Adding a script to add the package.json during building the toolchain would avoid manual work and errors.
I suggested a while ago #9

antmak, 2022-09-21:
thank you for pointing it out!

antmak, 2022-09-21:
@Jason2866 we've updated darwin_x86_64 and darwin_arm64 archives with fixed package.json-s, I hope it would work well if you uninstall and then install `espressif/toolchain-xtensa-esp32` package

Jason2866, 2022-09-21:
@antmak Platformio Github CI is running. Looks good for Linux /Windows MacOS (Intel). Can test for MacOS (arm) in a hour. I have no doubt it will be fine too.
Thx for the fast response! I will update here when Apple arm is working too.

closed 2022-09-25 by Jason2866