espressif/crosstool-NG #30: Please upload `esp32ulp-elf-v2.35_20220830` to Platformio Registry

Jason2866, 2022-09-25:
@antmak @igrr i know not the correct repo but the repo binutils-gdb has no issue section.
I did managed to do the needed changes in Platformio to use the new ulp toolchain.
Building the fsm and riscv ulp examples does work now too.

Would be really great if we (i would make the PR) could update Platformio development branch to the latest patch5 and a ulp toolchains. So all 'makers' can benefit of the bug fixes and the great performance gain when working with a Apple Silicon CPU.

Thx for your great work and support around your products.

antmak, 2022-09-28:
Hi @Jason2866,

The current IDF v4.4.2 still uses old ULP toolchains. But the coming (soon) _v4.4.3_ will get changed. The new ULP toolchain has [an option `--mcpu=`](https://github.com/espressif/esp-idf/commit/16dbb8f24963daac0d3ab931bd585f0910845093) to use with esp32/s2/s3 chips.
We uploaded [espressif/toolchain-esp32ulp](https://registry.platformio.org/tools/espressif/toolchain-esp32ulp), but IDF release/v4.4 branch haven't got support of the new ULP tools yet (in testing). 
Then PIO should be adapted then in some way

Jason2866, 2022-09-28:
@antmak  Thx, great! I have done the needed changes in Platformio to support the new stuff.
Currently i use our fork IDF 4.4.3 release. Hope the official one is ready soon :-)

closed 2022-09-28 by Jason2866