espressif/crosstool-NG #34: Consultation on crosstool license

fangye945a, 2022-11-09:
hello, every committer:
      I want to use crosstools([xtensa-esp32-elf-gcc8_4_0](https://github.com/espressif/crosstool-NG/releases/download/esp-2021r1/xtensa-esp32-elf-gcc8_4_0-esp-2021r1-linux-amd64.tar.gz
))  to build my project. Will the GPL license force me to open my source code? Will this affect the commercial use of my products?

igrr, 2022-11-09:
@fangye945a GCC runtime libraries (that is, libraries linked together with your application) have a special exception to the GPL license: https://www.gnu.org/licenses/gcc-exception-3.1.en.html
Therefore if you are simply compiling a program using GCC, it does not require you to comply with GPL.

fangye945a, 2022-11-09:
Thanks very much.

closed 2022-11-09 by fangye945a