espressif/crosstool-NG #36: Fix underlying gcc bug to support c++20 "requires"

simonmicro, 2022-12-06:
### Issue submission guidelines
See [detaled guidelines](http://crosstool-ng.github.io/support/). In short:
- Report crosstool-NG version: **As used by PlatformIO, I can only guess the value of `8.4.0+2021r2-patch5`**
- Report host OS and its version: **Linux Mint 21**
- Attach `.config` (unless the issue is not related to any configuration): **-**
- Attach `build.log` (unless crosstool-NG fails before the build starts): [build.log](https://github.com/espressif/crosstool-NG/files/10167458/build.log)

I'm very sorry, if this is the wrong repository to report this, but I have no idea where to report this instead. :sweat:

igrr, 2022-12-06:
Hi @simonmicro,
We have already released a toolchain based on GCC11, it is used in ESP-IDF v5.0 and later. This toolchain supports C++20 features, including `requires` expressions. 

Note that if you are using the Arduino-esp32 core (directly or via PlatformIO), you'll have to wait for some time until Arduino-esp32 upgrades to ESP-IDF v5.0 and starts using the GCC 11 based toolchain.

simonmicro, 2022-12-07:
Hi @igrr ,
Thank you very much for your explanation! I'll see if we could either remove the Arduino framework (unlikely), or we just have to wait for them to fix their stuff.
Thanks!

closed 2022-12-07 by simonmicro