espressif/crosstool-NG #43: esp-12.2.0_20230208 missing gdb

gudvinr, 2023-06-26:
Toolchain release **esp-12.2.0_20230208** is missing `esp-12.2.0_20230208`

igrr, 2023-06-26:
We are now releasing GDB separately from the rest of the toolchain, to allow more frequent updates. You can find GDB releases here: https://github.com/espressif/binutils-gdb/releases

gudvinr, 2023-06-26:
In 11.2 and earlier GDB was included in toolchain package and included in [toolchain-xtensa-esp32](https://registry.platformio.org/tools/espressif/toolchain-xtensa-esp32) platformio package too. I assume it's the same for esp32s2, esp32s3, etc.

But those packages now don't have gdb since 12.2 and there are no separate packages for gdb (at least not in the pio registry).

markand, 2024-02-09:
Hi,

I've faced the same issue, I was a bit surprised because crosstool-ng already builds binutils-gdb. Is there a way to make it build gdb as well? I've tried building espressif's fork of binutils-gdb with `--target xtensa-esp32s3-elf` as option but it builds more than gdb and replaced various toolchain files like `-as`, `-ld` and broke my toolchain. Trying to build a project for esp32s3 ended up with an error like "toolchain is little-endian but target is big-endian` which obviously looks wrong.

I know that building from source is not the official way of doing ESP development but sometimes may be needed for alternative platforms/libc's (e.g. musl on aarch64).

Can we can some directions on how to build just the gdb part for a single target in binutils-gdb fork?

Thanks.

Lapshin, 2024-06-10:
@markand , sorry for the long delay. I hope the comment https://github.com/espressif/crosstool-NG/issues/50#issuecomment-1942202443 can help you understand the changes after 12.2 release.

To build gdb you can refer to https://github.com/espressif/binutils-gdb repo ins it's [build script](https://github.com/espressif/binutils-gdb/blob/esp-gdb-14.2/build_esp_gdb.sh)

closed 2024-06-17 by Lapshin

shizacat, 2024-10-19:
@Lapshin,
Hello there! I would like to request that you could please add a link and description to the README file. I've been having a hard time figuring out why and where the gdb is located.
Thank you for your assistance!