espressif/crosstool-NG #46: xtensa-esp32s2-elf-gcc: internal compiler error: Segmentation fault signal terminated program cc1

briangshea, 2023-10-08:
### Issue submission guidelines
See [detaled guidelines](http://crosstool-ng.github.io/support/). In short:
- Report crosstool-NG version
- Report host OS and its version
- Attach `.config` (unless the issue is not related to any configuration)
- Attach `build.log` (unless crosstool-NG fails before the build starts)

I'm using the the compiler installed through platofomio.ini  with platform = espressif32@5.4.0
The same issues happen on 5.3.0, 5.2.0 and 5.1.0. have not tested prior to that.

I'm working on building from crosstool-ng 1.25 but have not been able to get platformio to accept my custom toolchain as of yet..

I have two systems, 1st AMD Ryzen 5950x w/ 64GB of ram, Linux Mint 20.3 with latest updates this is the system that builds always fail with.

I have an Intel Core I7 based Linux Mint 20.3 system with all the latest updates that everything works fine on. No seg faults.

The issue seems to happen at random almost never on the same file.

briangshea, 2023-10-16:
Good morning,

I was able to rebuild the crosstool-NG with tag esp-2021r2-patch5 and using xtensa-esp32s2-elf configuration on the AMD 5950x Linux mint 20.2 system, the seg fault no longer occurs. 

The only changes I made was to use a statically linked toolchain and pulled zlib from the git repo as the tar.gz fossil has been removed due to a bug in 1.2.12.

Attached is the configuration file i used for reference. 
NOTE: paths to store downloads and the build output folder have been modified

[config-xtensa-esp32s2-elf.tar.gz](https://github.com/espressif/crosstool-NG/files/12916833/config-xtensa-esp32s2-elf.tar.gz)

Lapshin, 2024-06-17:
Closing due to issue was outdated

closed 2024-06-17 by Lapshin