espressif/crosstool-NG #48: Support for C++20 ranges

LeCyberDucky, 2024-01-13:
Hey there!

Long story short, I'm looking for C++20 ranges support when programming my ESP32-S3 with the Arduino framework.

I'm working on a project for an ESP32-S3-based board (the LilyGo T-Display-S3), and, when trying to use the C++20 ranges library, I run into the error: 

`fatal error: ranges: No such file or directory`

I'm using PlatformIO to develop this project. An initial question in the [PlatformIO forum](https://community.platformio.org/t/c-20-ranges-library-fatal-error-ranges-no-such-file-or-directory/37771), lead me to the [ESP32 Arduino forum](https://esp32.com/viewtopic.php?f=19&t=37770), which has subsequently lead me to this place.

Along this adventure, I have come across this seemingly similar issue about support for C++20 chrono: https://github.com/espressif/esp-idf/issues/12914#issuecomment-1873703047

It is mentioned that this will be supported starting with GCC14. A quick look at https://en.cppreference.com/w/cpp/compiler_support reveals that "The One Ranges Proposal" should be supported from GCC10. And I can see that the latest release in this repository upgrades the GCC version to 13.2.0. So, I should be able to use ranges, no? 

All of this has me quite confused, especially considering that I'm simply trying to port a project that works on an ESP8266. Could somebody explain to me what is going wrong, and if there's anything I can do to fix it?

It is not quite obvious to me which compiler I'm actually using. 
My PlatformIO.ini looks like this: 
```
[env:lilygo-t-display-s3]
; platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git
board = lilygo-t-display-s3
framework = arduino
build_unflags = -std=gnu++11
build_flags = -std=gnu++2a //https://community.platformio.org/t/which-c-standard-am-i-using/24597/4
```
With that, I get the following information when trying to build my project:
```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```

Kind regards

igrr, 2024-01-13:
Looks like you are using Arduino-esp32 core version 2.0.14, which is based on ESP-IDF v4.x, which uses GCC 8.4.x for its toolchain. Newer IDF versions are already using the newer versions of GCC (12.2, 13.2). The upcoming Arduino-esp32 v3.0.0 release will be based on IDF v5.1.x, so the functionality you are looking for will be available there. Here is the project which tracks the 3.0.0 release of Arduino-esp32 core: https://github.com/orgs/espressif/projects/3/views/14

LeCyberDucky, 2024-01-13:
Thank you for the very quick and helpful reply! 

Whoa, that's quite a jump in the GCC versions. I gather that pre-releases of version 3.0.0 are available here: https://github.com/espressif/arduino-esp32/releases

Do you know whether it would be possible to change my setup to use that?

igrr, 2024-01-13:
Platformio does not yet have support for arduino-esp32 v3.0.0 pre-releases, however there is a fork maintained by Tasmota which already supports this version: https://github.com/Jason2866/platform-espressif32. Perhaps you can try it and see if that works.

LeCyberDucky, 2024-01-13:
Alright, that sounds great, I'll give it a try! Thanks for pointing me in the right direction. 

I suppose we can close this issue then :)

closed 2024-01-13 by igrr

LeCyberDucky, 2024-05-01:
@robertlipe I agree that it's very frustrating to deal with these layers upon layers of abstractions. 

I have not had the time to look at this for a little while now, so I'm a bit out of the loop and therefore not sure of how much help I can be. 

As I recall, however, I ended up making use of this: https://github.com/platformio/platform-espressif32/pull/1281, which let me compile the stuff I wanted to compile. Is that of any help?