espressif/crosstool-NG #49: fatal error: type_traits: No such file or directory

w2016561536, 2024-02-10:
When i tried to compile for esp32s3, the c++ headers file <type_traits> could not be found. However cmath, cstdio, cstring could be found normally.

Compiler version: esp-13.2.0_20230928
OS: Ubuntu 22.04.3 LTS on 5.15.133.1-microsoft-standard-WSL2

Compile command: 
` /home/wyt/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -DCONFIG_ARCH_BOARD_PX4_ESP32S3 -DCONSTRAINED_MEMORY -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I/home/wyt/px4_14_esp32/PX4-Autopilot/boards/px4/esp32s3/src -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/src/px4/common/include -I/home/wyt/px4_14_esp32/PX4-Autopilot/build/px4_esp32s3_default -I/home/wyt/px4_14_esp32/PX4-Autopilot/build/px4_esp32s3_default/src/lib -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/src/px4/espressif/esp32s3/include -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/common -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/common/include -I/home/wyt/px4_14_esp32/PX4-Autopilot/src -I/home/wyt/px4_14_esp32/PX4-Autopilot/src/include -I/home/wyt/px4_14_esp32/PX4-Autopilot/src/lib -I/home/wyt/px4_14_esp32/PX4-Autopilot/src/lib/matrix -I/home/wyt/px4_14_esp32/PX4-Autopilot/src/modules -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/xtensa/src -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/xtensa/src/chip -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/xtensa/src/common -I/home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/NuttX/apps/include -isystem /home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/NuttX/include/cxx -isystem /home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/include/cxx -isystem /home/wyt/px4_14_esp32/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/include -mlongcalls -Wno-format -Wno-unused-function -Os -DNDEBUG -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Wno-address -Wno-error=dangling-pointer -Wno-error=array-bounds -Wno-error=format-overflow= -Wno-error=unused-variable -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -Wreorder -Wno-overloaded-virtual -fno-rtti -fno-exceptions -fno-sized-deallocation -fno-threadsafe-statics -nostdinc++ -std=gnu++14 -MD -MT msg/CMakeFiles/uorb_msgs.dir/topics_sources/autotune_attitude_control_status.cpp.obj -MF msg/CMakeFiles/uorb_msgs.dir/topics_sources/autotune_attitude_control_status.cpp.obj.d -o msg/CMakeFiles/uorb_msgs.dir/topics_sources/autotune_attitude_control_status.cpp.obj -c /home/wyt/px4_14_esp32/PX4-Autopilot/build/px4_esp32s3_default/msg/topics_sources/autotune_attitude_control_status.cpp `

Related header file: 
[matrix.txt](https://github.com/espressif/crosstool-NG/files/14229801/matrix.txt)

igrr, 2024-02-10:
I think the issue is that there is `-nostdinc++` on the command line, so `<type_traits>` doesn't get found.

If you remove `-nostdinc++`, the header should get found correctly: https://godbolt.org/z/dKTfTTqvK

w2016561536, 2024-02-11:
> I think the issue is that there is `-nostdinc++` on the command line, so `<type_traits>` doesn't get found.
> 
> If you remove `-nostdinc++`, the header should get found correctly: https://godbolt.org/z/dKTfTTqvK

Hi @igrr , It works! Thanks.

closed 2024-02-11 by w2016561536