espressif/crosstool-NG #50: xtensa-esp-elf: how to influence endianness ?

gopakumarce, 2024-02-12:
[Posted here too: https://github.com/esp-rs/esp-idf-sys/issues/271]

In esp-idf version 5.1 onwards, the gcc used is just xtensa-esp-elf - that gcc has __XCHAL_HAVE_BE set to 1 .. So if I am compiling for esp32, what exactly is the magic going on behind the scenes to compile as little endian ? xtensa-esp32-elf has it set to 0 nicely. I see a lot of discussions online which says an "xtensa-overlays.h" or "xtensa-config.h" or "core.h" or "core-isa.h" can be used to influence gcc to produce BE/LE, but I cant find exactly how that works - any pointers will be highly appreciated

@Lapshin I see the commit 96914125ba6feb10e6943136160bddea759a2b6f where you had unified the toolchain - I am guessing you are probably the only one who can help with an answer to this question.

Lapshin, 2024-02-13:
Hi @gopakumarce !

Before GCC release 13 we had to build toolchain for every chip. Before building we just replaced files in binutils/gcc/newlib from this repository [xtensa-overlays](https://github.com/espressif/xtensa-overlays/tree/master). It contains all chip-specific information.

After GCC 13 is released the repository [xtensa-dynconfig](https://github.com/jcmvbkbc/xtensa-dynconfig) used to build chip-specific dynamic libraries that contains information from [xtensa-overlays](https://github.com/espressif/xtensa-overlays/tree/master). (You can find libraries `xtensa_esp*.so` in the latest toolchain distribution).

Also, for backward compatibility gcc-wrappers `xtensa-[esp32|esp32s2|esp32s3]-elf-*` exists in the distro. When executing they just append your command line with `-mdynconfig=xtensa_[esp32|esp32s2|esp32s3].so`. And also sets env variable `XTENSA_GNU_CONFIG`.
Note: in general set `XTENSA_GNU_CONFIG` is enough to use the right chip-specific configuration. `-mdynconfig` option needs to find the right `libc` for this chip in multilibs.

Hope this helps you. If you still have any questions, I'd be happy to answer them.

gopakumarce, 2024-02-13:
> Hi @gopakumarce !
>
> Before GCC release 13 we had to build toolchain for every chip. Before building we just replaced files in binutils/gcc/newlib from this repository [...]

Thanks a ton @Lapshin for the response, that would have taken me a long time to figure out. So I was searching for where in the latest esp-idf (makefiles/cmakefiles) do they set XTENSA_GNU_CONFIG or pass -mdynconfig and I cant find it, do you know if they are already using xtensa-esp-elf or they are still using the wrappers ?

Lapshin, 2024-02-13:
@gopakumarce , it using wrappers. for example see this file: https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-esp32s3.cmake#L5-L8

gopakumarce, 2024-02-14:
> @gopakumarce , it using wrappers. for example see this file: https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-esp32s3.cmake#L5-L8

Thx again @Lapshin

closed 2024-02-14 by gopakumarce