espressif/crosstool-NG #51: 我使用Arduion编译Esp32项目时出错,换用2.0.11\2.0.12

BalancedTernary, 2024-02-20:
### Issue submission guidelines
See [detaled guidelines](http://crosstool-ng.github.io/support/). In short:
- Report crosstool-NG version
- Report host OS and its version
- Attach `.config` (unless the issue is not related to any configuration)
- Attach `build.log` (unless crosstool-NG fails before the build starts)

BalancedTernary, 2024-02-20:
sorry,我误触了回车键

Lapshin, 2024-06-10:
@BalancedTernary , [the toolchain release](https://github.com/espressif/crosstool-NG/releases/tag/esp-13.2.0_20240530) has fixes for paths on windows. Could you give it a try?

BalancedTernary, 2024-06-21:
> @BalancedTernary , [the toolchain release](https://github.com/espressif/crosstool-NG/releases/tag/esp-13.2.0_20240530) has fixes for paths on windows. Could you give it a try?

I currently alleviate this problem by manually copying files to the lookup directory. When I need to configure the environment, I try it. But I'm afraid to tweak the environment that's already available. Sorry~