espressif/crosstool-NG #57: gold / lld support for parallel linking

rojer, 2024-05-29:
Linking phase is taking substantial amount of time on larger projects, it would be great to get support for `-fuse-ld=gold` and/or `-fuse-ld=lld`.

rojer, 2024-05-29:
@igrr how feasible is this?

i took a quick stab at `gold` support and found:
* `--dynconfig` is not supported - not sure how big of a deal this is at linking stage
* `--no-warn-rwx-segments` is not supported - probably minor
* `REGION_ALIAS` linker script function is not supported. a bit annoying, but can be worked around with some linker script changes.

after that i ran into "unsupported ELF machine" error (i was using an amd64 build of gold, so not surpirsing)

Lapshin, 2024-06-10:
@rojer , gold has no xtensa support yet. It could not process required xtensa-specific relocations until not implemented. It will likely not get it due to very low priority.

rojer, 2024-06-10:
ok, what about `lld`? for our app, it's not uncommon for linking to take 5-15 seconds (depending on hardware). fast incremental builds (1 source file change) are important for developer productivity and currently linking is the biggest factor here.

Lapshin, 2024-06-10:
@rojer , you can refer to LLVM project (but it is not stable yet). Seems LLD is supported for versions greater than [esp-15.0.0-20221201](https://github.com/espressif/llvm-project/releases/tag/esp-15.0.0-20221201). (See https://github.com/espressif/llvm-project/pull/53)

rojer, 2024-06-10:
ok, does this mean that it can be integrated into IDF as an option, or even the default? i'm looking at the toolchain distribution archive and not sure if the xtensa or resv-v variant is included, i only see amd64 and wasm variants (`ld64.lld` and `wasm-ld`):

```
[rojer@nbf ~/allterco/esp-idf 5.2.1-s2]$ tar tvf llvm-esp-16.0.0-20230516-linux-amd64.tar.xz | grep 'lld'
lrwxrwxrwx root/root         0 2023-05-17 10:58 esp-clang/bin/ld.lld -> lld
-rwxr-xr-x root/root   5296448 2023-05-17 10:58 esp-clang/bin/lld
lrwxrwxrwx root/root         0 2023-05-17 10:58 esp-clang/bin/lld-link -> lld
lrwxrwxrwx root/root         0 2023-05-17 10:58 esp-clang/bin/wasm-ld -> lld
lrwxrwxrwx root/root         0 2023-05-17 10:58 esp-clang/bin/ld64.lld -> lld
[rojer@nbf ~/allterco/esp-idf 5.2.1-s2]$ tar tvf clang-esp-17.0.1_20240419-x86_64-linux-gnu.tar.xz | grep 'lld'
lrwxrwxrwx root/root         0 2024-04-19 23:27 esp-clang/bin/lld-link -> lld
lrwxrwxrwx root/root         0 2024-04-19 23:27 esp-clang/bin/ld.lld -> lld
-rwxr-xr-x root/root  67654984 2024-04-19 22:54 esp-clang/bin/lld
lrwxrwxrwx root/root         0 2024-04-19 23:27 esp-clang/bin/ld64.lld -> lld
lrwxrwxrwx root/root         0 2024-04-19 23:27 esp-clang/bin/wasm-ld -> lld
```

Lapshin, 2024-06-11:
@gerekon PTAL

gerekon, 2024-06-12:
@rojer LLD supplied with Espressif Clang toolchain supports Xtensa and RISCV archs, but it has some issues.
1) For Xtensa it includes preliminary support and still has some issues with building IDF apps.
2) RISCV support comes from the upstream, but again there are some issues with building IDF apps.

Currently we are working on fixing IDF builds with LLD for RISCV chips. Hopefully it will get into the next release.
After that we will start working on issues related to IDF builds with LLD for Xtensa chips.

gerekon, 2024-06-12:
@rojer But note, that Xtensa support in Clang itself is not optimal as of now (code size and perf), e.g. direct calls are not implemented yet. https://github.com/espressif/llvm-project/issues/90. So it seems to be early to switch to Clang toolchain completely. May be we can find a way to use LLD with GCC as an option.

rojer, 2024-06-12:
oh yes, clang is hopeless in `-Os` mode, 30% bigger code or so last i checked.
what i was hoping for was to compile gcc and link with lld.

ok, thank you for the update, please know that this is eagerly awaited :)