espressif/crosstool-NG #61: Support for Windows ARM64

Mlebailly22, 2024-07-21:
Hi, 

With the expension of Windows ARM computer, Is it possible to have the ARM version of "toolchain-xtensa-esp32s3",

Because I get this error on platformIO  on VSC, and I don't know what ca be the solution : 
[image]

Thanks for your help

Lapshin, 2024-07-30:
@Mlebailly22 , thank you for your patience. Currently, we don't have toolchain builds for windows-arm.

Now you can try to use Windows Subsystem for Linux (WSL). There are related links that may help you with:

- https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#windows-10
- https://code.visualstudio.com/docs/remote/wsl

Jason2866, 2024-08-06:
@Mlebailly22 Doing this suggested changes in your local Platformio should make it possible to compile successfully.
https://github.com/platformio/platformio-core/pull/4952

Mlebailly22, 2024-08-11:
Hi @Jason2866 ,
Yes Actualy, this PR have been create because I also ask the same question on another forum
[https://community.platformio.org/t/windows-on-arm64-problem-installing-xtensa-toolchain/25497/10?u=melvinlebailly22](url)
But I didn't know that I can do these changes on my local Platformio 😅
I have no idea where I can find the file util.py, and after modifying it, Where I need to add this PLATFORMIO_SYSTEM_TYPE,

Thanks you very much for your help

Jason2866, 2024-08-11:
Adding `PLATFORMIO_SYSTEM_TYPE = windows_amd64` in your Windows env should do.
Random Google hit for setting an Windows env https://www.computerhope.com/issues/ch000549.htm

Mlebailly22, 2024-08-11:
> Changes in your local Platformio

Okay for the Windows env, and what about Finding the util.py file ?

Thanks

Jason2866, 2024-08-11:
There is a thing called Internet search ;-)
https://www.lifewire.com/11-free-file-search-tools-1356644

closed 2024-10-15 by Lapshin