espressif/crosstool-NG #62: localtime() always returns 23:59 in release: v12.2.0+20230208

mahyar1284, 2024-08-05:
- Toolchain Version:  12.2.0+20230208
- Host OS: Windows 11 64bit
- platform and framework: esp32 dev module Platfromio | Arduino
- Sample code to recreate the issue:

```c++
    // Set the desired time (4:20)
    int desiredHours = 4;
    int desiredMinutes = 20;

    // Create a struct timeval
    struct timeval newTime;
    newTime.tv_sec = 0; // Initialize seconds (will be set later)
    newTime.tv_usec = 0; // Initialize microseconds (not used here)

    // Get the current time
    time_t currentTime;
    time(&currentTime);

    // Convert to a readable format
    struct tm* localTime = localtime(&currentTime);
    localTime->tm_hour = desiredHours;
    localTime->tm_min = desiredMinutes;
    localTime->tm_isdst = -1; // Set DST to be determined automatically

    // Set the seconds in the timeval struct
    newTime.tv_sec = mktime(localTime);

    // Set the system time
    int rc = settimeofday(&newTime, NULL);
    if (rc == 0) {
        printf("System time set to %02d:%02d\n", desiredHours, desiredMinutes);

        // Now let's read the updated time
        struct tm* updatedLocalTime = localtime(&newTime.tv_sec);
        int updatedHours = updatedLocalTime->tm_hour;
        int updatedMinutes = updatedLocalTime->tm_min;

        printf("Updated time (read using localtime()): %02d:%02d\n", updatedHours, updatedMinutes);
    } else {
        perror("Error setting system time");
    }
```
the above code works fine with toolchain version 8.4.0

Lapshin, 2024-08-07:
@mahyar1284 , the root cause of the issue you found is that `struct tm localTime;` allocated on stack and was not initialized. I suspect the difference between 12.2.0 and 8.4.0 toolchains that your `localTime` contains invalid and valid garbage inside... And since you do not check `mktime(&localTime)` return value, `settimeofday` set value that you did not expect.

You may fix your code just with initialized `struct tm localTime = {0};`

Also, it would be better if you checked all the return codes in your applications.

mahyar1284, 2024-08-08:
@Lapshin thanks for your advice. I initialized it like this  `struct tm localTime = {0};` and now instead of returning constantly 23:59 (which not initializing localTime caused mktime to return -1 and therefore a 23:59 output) it returns random time and date. I wanted to find the root cause of this issue so instead of using mktime and these stuff, I decided to set epoch value and read the epoch directly without converting it to a human readable format.

```c
struct timeval tv = { .tv_sec = 1723118952, .tv_usec = 0 };
settimeofday(&tv, NULL);
while(true)
{
     time_t now;
     time(&now);
     printf("Current epoch time: %ld\n", now);
     delay(500);
}
```
I set the system time epoch to `1723118952` and the read it using `time(&now)` and print it. the initial printed value is 61 or 62 randomly but afterwards it switches between 996 and 500996 periodically. 

serial monitor output:
```
Current epoch time: 61
Current epoch time: 996
Current epoch time: 500996
Current epoch time: 996
Current epoch time: 500996
Current epoch time: 996
Current epoch time: 500996
```

Lapshin, 2024-08-14:
@mahyar1284 ,  I could not reproduce it with IDF release/v5.1 that uses `esp-12.2.0_20230208` toolchain.

code:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <time.h>

void app_main(void)
{
    struct timeval tv = { .tv_sec = 25, .tv_usec = 0 };
    _gettimeofday_r(__getreent(), &tv, NULL);
    while(true)
    {
        time_t now;
        time(&now);
        printf("Current epoch time: %lld\n", now);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```

Output:

```
...
I (324) main_task: Started on CPU0
I (334) main_task: Calling app_main()
Current epoch time: 0
Current epoch time: 1
Current epoch time: 2
Current epoch time: 3
Current epoch time: 4
Current epoch time: 5
Current epoch time: 6
Current epoch time: 7
.......
```

You may try to ask in arduino project

mahyar1284, 2024-08-20:
@Lapshin thanks for testing my code 😊🙏. That's odd, I though the `time(&now);` function is implemented inside toolchain and it is not related to arduino. by the way I'm curious where is the implementations of `time(&now);` and `settimeofday();` functions. can u help me where can I find them?

Lapshin, 2024-08-21:
@mahyar1284 , time() indeed located in libc. But in our case it's just a wrapper, see the code: https://github.com/espressif/newlib-esp32/blob/7cdcb0b0848a5d909e906834fe81903ba9466c90/newlib/libc/time/time.c#L35-L48

It calls `_gettimeofday_r` that defined in IDF here https://github.com/espressif/esp-idf/blob/dbce23f8a449eb436b0b574726fe6ce9a6df67cc/components/newlib/time.c#L167-L182

Sorry, I don't know how arduino works and even where is their source code located...

igrr, 2024-08-21:
Arduino core for ESP32 (https://github.com/espressif/arduino-esp32) is built on top of ESP-IDF, so when you call `time` function in an Arduino sketch, the implementation from IDF is used.

Lapshin, 2024-08-21:
@mahyar1284 , the significant change after upgrade to GCC 11+ is `time_t` become 64bit instead 32bit. https://github.com/espressif/esp-idf/issues/584

I hope all will work correctly if you make a change in your code:

```diff
-printf("Current epoch time: %ld\n", now);
+printf("Current epoch time: %lld\n", now);
```

The problem is that arduino compile the code without `-Wall` flag that's why the behaviour was misunderstood. IDF uses the flag by default and if compile your code it gives something like:

```
/home/alex/git/esp-idf/examples/get-started/hello_world/main/hello_world_main.c:24:32: error: format '%d' expects argument of type 'int', but argument 5 has type 'time_t' {aka 'long long int'} [-Werror=format=]
   24 |         printf("%u %u %d %d\n", sizeof(time_t), sizeof(long), __LONG_MAX__ > 0x7fffffffL, now);
      |                          ~^                                                               ~~~
      |                           |                                                               |
      |                           int                                                             time_t {aka long long int}
      |                           %lld

```

Lapshin, 2024-08-22:
@mahyar1284 , arduino don't enable compile warnings by default. You may enable it in "file"->"preferences"->"Compiler warnings".Closing this issue as answered.

closed 2024-08-22 by Lapshin