espressif/crosstool-NG #64: Question: Benefit of using picolib

Jason2866, 2024-09-02:
don't know where else to ask.
@Lapshin What benefits are seen by using picolib? Smaller code size?

Lapshin, 2024-09-02:
Hi @Jason2866 ,  yes, that is it! Moreover, the benefit is related only to printf-family functions. That is because picolib is based on newlib's sources. Picolibc also has some changes in the math library, but I did not test it well. I did not find many differences after briefly looking at the Picolibc math sources. 

The new release is coming soon and will include newlib and picolibc libraries. So, you will able to switch between libc implementations using IDF configure option

Jason2866, 2024-09-04:
The repo https://github.com/espressif/picolibc is not public. Will this be changed?

Jason2866, 2024-10-03:
@Lapshin Thx!

closed 2024-10-03 by Jason2866