espressif/crosstool-NG #65: esp32 vs esp when building from source

leres, 2024-09-27:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I've been using 1.22.0.g20200907 built from source on FreeBSD/amd64 for a number of years. In this case the compiler binary that is built is named xtensa-esp32-elf-cc.

I started looking at using a newer idf which lead me to building esp-13.2.0_20240530 and much confusion. The samples/xtensa-esp32-elf crosstool.config specifies gcc 12.2.0 instead of 13.2.0. But when I download and unpack the linux-gnu tarball I find:
- xtensa-esp32-elf-cc
- xtensa-esp32s2-elf-cc
- xtensa-esp32s3-elf-cc

which are all identical.

I've built esp-13.2.0_20240530 using the xtensa-esp-elf crosstool.config and it produces xtensa-esp-elf-cc, perhaps expected given the name of the samples directory (and CT_TARGET_VENDOR in the config).

I also looked at esp-14.2.0_20240906 and it only has the xtensa-esp-elf crosstool.config, the others were removed via 6269489 -- "samples: remove obsolete xtensa-esp32XX-elf targets"

If I want to build a 13.2.0 cross compiler that targets esp32, should I use the xtensa-esp-elf crosstools.config? Or do I need to create a custom crosstools.config that sets CT_TARGET_VENDOR to esp32?

I am very confused.

Lapshin, 2024-10-03:
@leres , since gcc version 12 xtensa cpu can be set with using `XTENSA_GNU_CONFIG` env variable that points to xtensa_dynconfig_library.so file (See https://github.com/gcc-mirror/gcc/commit/ecb575d09c0ff5314088214c2bb39f1959ad3318)

After this change the toolchain's target triplet is `xtensa-esp-elf`. Other files started with `xtensa-esp32XX-...` are just binary wrappers. See source code https://github.com/espressif/esp-toolchain-bin-wrappers/blob/62776e937e4fb6e9d0ff6081a5ca9571b210f7ba/gnu-xtensa-toolchian/main.rs. Build script for this you can find here https://github.com/espressif/crosstool-NG/blob/esp-1.26.x/.gitlab-ci.yml#L115-L141

Also, instead of building binary wrappers you may just copy them from existing release (e.g.: https://github.com/espressif/crosstool-NG/releases/tag/esp-14.2.0_20240906)

BTW, why do you need to build cross-compiler with GCC 13.2.0 ? I'm asking because it already could be found here https://github.com/espressif/crosstool-NG/releases/tag/esp-13.2.0_20240530

Lapshin, 2024-10-03:
@leres , also you could find some details here https://github.com/espressif/crosstool-NG/issues/50#issuecomment-1942202443

leres, 2024-10-16:
> @leres , since gcc version 12 xtensa cpu can be set with using `XTENSA_GNU_CONFIG` env variable that points to xtensa_dynconfig_library.so file (See [gcc-mirror/gcc@ecb575d](https://github.com/gcc-mirror/gcc/commit/ecb575d09c0ff5314088214c2bb39f1959ad3318))
> 
> After this change the toolchain's target triplet is `xtensa-esp-elf`. Other files started with `xtensa-esp32XX-...` are just binary wrappers. See source code https://github.com/espressif/esp-toolchain-bin-wrappers/blob/62776e937e4fb6e9d0ff6081a5ca9571b210f7ba/gnu-xtensa-toolchian/main.rs. Build script for this you can find here https://github.com/espressif/crosstool-NG/blob/esp-1.26.x/.gitlab-ci.yml#L115-L141

Thank you. Super helpful. I now have something that works.

> BTW, why do you need to build cross-compiler with GCC 13.2.0 ? I'm asking because it already could be found here https://github.com/espressif/crosstool-NG/releases/tag/esp-13.2.0_20240530

I'm a FreeBSD developer while I could run linux binaries in compatibility mode I wanted a native toolchain.

closed 2024-10-16 by leres