espressif/crosstool-NG #68: CMake: xtensa-esp32s3-elf-gcc is not able to compile a simple test program

yh-sb, 2024-11-26:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

esp32-elf toolchain is able to work fine and compile a simple test program

### Actual behavior (suspected bug)

Got the following error at the very beginning when trying to compile simple project.
`xtensa-esp32s3-elf-gcc is not able to compile a simple test program`

**NOTE:** there is no such error when using latest esp32-elf toolchain (_13.2.0_20240530_) from [ESP-IDF v5.3.1 package](https://dl.espressif.com/dl/esp-idf). Everything compiles fine in this case.

### Error logs or terminal output

```shell
cmake . -Bbuild -G Ninja -DCMAKE_BUILD_TYPE=Debug
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/mydevtools/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files/mydevtools/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/mydevtools/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - broken
CMake Error at C:/Program Files/mydevtools/CMake/share/cmake-3.30/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "C:/Program Files/mydevtools/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: 'D:/dev/cpp/mcu-cpp/build/CMakeFiles/CMakeScratch/TryCompile-851lr3'

    Run Build Command(s): C:/PROGRA~1/MYDEVT~1/Ninja/ninja.exe -v cmTC_e1922
    [1/2] C:\PROGRA~1\MYDEVT~1\XTENSA~2\bin\XT34AB~1.EXE    -o CMakeFiles/cmTC_e1922.dir/testCCompiler.c.obj -c D:/dev/cpp/mcu-cpp/build/CMakeFiles/CMakeScratch/TryCompile-851lr3/testCCompiler.c
    FAILED: CMakeFiles/cmTC_e1922.dir/testCCompiler.c.obj
    C:\PROGRA~1\MYDEVT~1\XTENSA~2\bin\XT34AB~1.EXE    -o CMakeFiles/cmTC_e1922.dir/testCCompiler.c.obj -c D:/dev/cpp/mcu-cpp/build/CMakeFiles/CMakeScratch/TryCompile-851lr3/testCCompiler.c
    cc1.exe: fatal error: Both 'XTENSA_GNU_CONFIG' and "-dynconfig=" specified but pointed different files: "'C:\PROGRA~1\MYDEVT~1\XTENSA~2\lib\XTENSA~3.SO'" "'xtensa_esp32s3.so'"
    compilation terminated.
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:12 (project)

-- Configuring incomplete, errors occurred!
```

### Steps to reproduce the behavior

```sh
cmake . -Bbuild -G Ninja -DCMAKE_BUILD_TYPE=Debug
cmake --build build -j
```

<details>
<summary>CMakeLists.txt</summary>

```cmake
cmake_minimum_required(VERSION 3.15)

find_program(CMAKE_C_COMPILER NAMES xtensa-esp32s3-elf-gcc)
find_program(CMAKE_CXX_COMPILER NAMES xtensa-esp32s3-elf-g++)
find_program(CMAKE_ASM_COMPILER NAMES xtensa-esp32s3-elf-gcc)
find_program(CMAKE_OBJCOPY NAMES xtensa-esp32s3-elf-objcopy)
find_program(CMAKE_SIZE NAMES xtensa-esp32s3-elf-size)
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR esp32)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)

project(esp32s3 C CXX ASM)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
set(CMAKE_CXX_STANDARD 20)

add_compile_options(
    $<$<NOT:$<COMPILE_LANGUAGE:ASM>>:-mlongcalls>
    -ffunction-sections
    -fdata-sections
    $<$<COMPILE_LANGUAGE:CXX>:-fno-exceptions>
    $<$<COMPILE_LANGUAGE:CXX>:-fno-rtti>
    $<$<COMPILE_LANGUAGE:CXX>:-fno-threadsafe-statics>
    $<$<COMPILE_LANGUAGE:CXX>:-fno-use-cxa-atexit>
)

add_executable(${CMAKE_PROJECT_NAME} main.cpp)

# esp-idf
include(third_party/esp-idf/tools/cmake/idf.cmake)
idf_build_process(esp32s3
    COMPONENTS freertos nvs_flash esptool_py esp_wifi
    SDKCONFIG_DEFAULTS "${CMAKE_CURRENT_LIST_DIR}/sdkconfig.defaults"
    SDKCONFIG "${CMAKE_BINARY_DIR}/sdkconfig"
)
idf_build_executable(${CMAKE_PROJECT_NAME})

target_link_options(${CMAKE_PROJECT_NAME} PRIVATE
    -nostdlib
    -Wl,--gc-sections
    -Wl,-Map=${CMAKE_PROJECT_NAME}.map,--cref
)

target_link_libraries(${CMAKE_PROJECT_NAME} PRIVATE
    idf::freertos
    idf::nvs_flash
    idf::esp_wifi
)
```

</details>

<details>
<summary>main.cpp</summary>

```cpp
// Example for ESP32-S3

#include <cstring>
#include "FreeRTOS.h"
#include "task.h"
#include "esp_system.h"
#include "esp_mac.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "nvs_flash.h"

static void heartbeat_task(void *pvParameters)
{
    while(1)
    {
        // Toggle the LED
        
        vTaskDelay(pdMS_TO_TICKS(500));
    }
}

static void wifi_event_handler(void *ctx, esp_event_base_t event_base,
    int32_t event_id, void* event_data)
{
    printf("wifi_event_handler() event_id=%d\n", event_id);
    
    switch(event_id)
    {
        case WIFI_EVENT_AP_STACONNECTED:
        {
            wifi_event_ap_staconnected_t *event = static_cast<wifi_event_ap_staconnected_t *>(event_data);
            printf("station " MACSTR " join, AID=%d\n", MAC2STR(event->mac), event->aid);
        }
        break;
        
        case WIFI_EVENT_AP_STADISCONNECTED:
        {
            wifi_event_ap_stadisconnected_t *event = static_cast<wifi_event_ap_stadisconnected_t *>(event_data);
            printf("station " MACSTR " leave, AID=%d\n", MAC2STR(event->mac), event->aid);
        }
        break;
    }
}

static void wifi_init_softap(void)
{
    esp_netif_init();
    
    esp_event_loop_create_default();
    esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, nullptr);
    
    wifi_init_config_t wifi_init_cfg = WIFI_INIT_CONFIG_DEFAULT();
    esp_wifi_init(&wifi_init_cfg);
    esp_wifi_set_storage(WIFI_STORAGE_RAM);
    
    wifi_config_t wifi_cfg = {};
    wifi_cfg.ap.max_connection = 4;
    strcpy((char *)wifi_cfg.ap.ssid, "esp32-ap2");
    wifi_cfg.ap.ssid_len = sizeof("esp32-ap2") - 1;
    strcpy((char *)wifi_cfg.ap.password, "wifipass");
    wifi_cfg.ap.authmode = WIFI_AUTH_OPEN;
    
    esp_wifi_set_mode(WIFI_MODE_AP);
    esp_wifi_set_config(WIFI_IF_AP, &wifi_cfg);
    esp_wifi_start();
}

// uart baud rate is 115200
extern "C" void app_main()
{
    esp_err_t res = nvs_flash_init();
    if (res == ESP_ERR_NVS_NO_FREE_PAGES || res == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        nvs_flash_erase();
        res = nvs_flash_init();
    }
    wifi_init_softap();
    
    xTaskCreate(heartbeat_task, "heartbeat", configMINIMAL_STACK_SIZE, nullptr, 1, nullptr);
}
```

</details>

esp-idf v5.3.1

### Project release version

esp-14.2.0_20241119, esp-13.2.0_20240530

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11 23H2 22631.4460

### Shell

CMD

### Additional context

There is no such error when using latest esp32-elf toolchain (_13.2.0_20240530_) from [ESP-IDF v5.3.1 package](https://dl.espressif.com/dl/esp-idf). Everything compiles fine in this case.

Lapshin, 2024-11-27:
@yh-sb , thank you for the report. I don't see the differences in a code between `13.2.0_20240530` and later releases. All of them are affected by such behavior.

Could you please move the toolchain to a path without spaces? It seems that this is the issue. Spaces in paths are not yet supported for Windows in IDF and related tools (yet). [IDF Windows installer](https://dl.espressif.com/dl/esp-idf/) doesn't even allow installation to a path with spaces.

yh-sb, 2024-11-27:
Yeah @Lapshin, you are indeed right. Just checked it out.

But, interesting that toolchain with "hotfix" works fine with spaces in path:
[xtensa-esp-elf-13.2.0_20240530-x86_64-w64-mingw32_hotfix.zip](https://github.com/espressif/crosstool-NG/releases/download/esp-13.2.0_20240530/xtensa-esp-elf-13.2.0_20240530-x86_64-w64-mingw32_hotfix.zip)

Lapshin, 2024-12-30:
@yh-sb , spaces in windows paths will be supported and tested in the next release

closed 2025-06-20 by yh-sb