espressif/crosstool-NG #69: _Pragma warning weird interaction

KaeLL, 2024-12-06:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Compiling one of my apps against the latest IDF:master, the following code
[image]
[image]
[image]

when compiled, produces the following warnings
```yaml
/app/components/w5100/port/src/w5100-ll.c: In function 'w5100_spi_init':
/app/components/w5100/port/src/w5100-ll.c:56:27: warning: VSPI_HOST is deprecated in favor of SPI3_HOST
   56 |                 &w5100_spi_handle ) );
      |                           ^~~~~~~~~~~~
[994/1023] Building C object esp-idf/nfc/CMakeFiles/__idf_nfc.dir/rfid_reader/Hardware/spi_nfc.c.obj
/app/components/nfc/rfid_reader/Hardware/spi_nfc.c: In function 'SPI_setup':
/app/components/nfc/rfid_reader/Hardware/spi_nfc.c:379:27: warning: VSPI_HOST is deprecated in favor of SPI3_HOST
  379 |                 &trf_hdl ) );
      |                           ^~~
[1015/1023] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/app/main/main.c: In function 'app_main':
/app/main/main.c:61:34: warning: VSPI_HOST is deprecated in favor of SPI3_HOST
   61 |                         SPI_DMA_CH1 ) );
      |                                  ^~~~~~~
```
That looks a bit weird. The lines reported point to the last line of the ESP_ERROR_CHECK macro calls, not the first. The markers are a bit off as well.
Then I checked the macros
```c
#define SPI_HOST    _Pragma ("GCC warning \"SPI_HOST is deprecated in favor of SPI1_HOST\"")  SPI1_HOST
#define HSPI_HOST   _Pragma ("GCC warning \"HSPI_HOST is deprecated in favor of SPI2_HOST\"") SPI2_HOST
#define VSPI_HOST   _Pragma ("GCC warning \"VSPI_HOST is deprecated in favor of SPI3_HOST\"") SPI3_HOST
```
I guess C99 _Pragma's don't interact well with the diagnostics part of the front-end? Either way, looks like a bug.

Lapshin, 2024-12-07:
@KaeLL , thank you for the report!

Could you please attach minimal code reproducer in text (not pictures)? Along with command line you compile it

KaeLL, 2024-12-07:
create main.c, paste and compile this
```c
#include "driver/spi_master.h"

void app_main( void )
{
	ESP_ERROR_CHECK( spi_bus_initialize(
		VSPI_HOST,
		&( const spi_bus_config_t ) {
			.miso_io_num = 19,
			.mosi_io_num = 23,
			.sclk_io_num = 18,
			.quadwp_io_num = -1,
			.quadhd_io_num = -1,
			.isr_cpu_id = ESP_INTR_CPU_AFFINITY_1,
			.intr_flags = ESP_INTR_FLAG_IRAM },
		SPI_DMA_CH1 ) );

	spi_device_handle_t trf_hdl, w5100_spi_handle;
	ESP_ERROR_CHECK( spi_bus_add_device(
		VSPI_HOST,
		&( const spi_device_interface_config_t ) {
			.command_bits = 3,
			.address_bits = 5,
			.spics_io_num = 26,
			.clock_speed_hz = SPI_MASTER_FREQ_20M / 4,
			.mode = 1,
			.queue_size = 1,
			.cs_ena_pretrans = 1,
		},
		&trf_hdl ) );

	ESP_ERROR_CHECK( spi_bus_add_device(
		VSPI_HOST,
		&( const spi_device_interface_config_t ) {
			.clock_speed_hz = 1200000,
			.spics_io_num = 17,
			.queue_size = 1,
			.pre_cb = NULL,
			.post_cb = NULL },
		&w5100_spi_handle ) );
}
```
It'll get you this
```yaml
[4/9] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/esp_gcc_69.c.obj
/esp_gcc_69/main/esp_gcc_69.c: In function 'app_main':
/esp_gcc_69/main/esp_gcc_69.c:16:27: warning: VSPI_HOST is deprecated in favor of SPI3_HOST
   16 |                 SPI_DMA_CH1 ) );
      |                           ^~~~~~                                         
/esp_gcc_69/main/esp_gcc_69.c:29:27: warning: VSPI_HOST is deprecated in favor of SPI3_HOST
   29 |                 }, &trf_hdl ) );
      |                           ^~~~~~                                         
/esp_gcc_69/main/esp_gcc_69.c:39:27: warning: VSPI_HOST is deprecated in favor of SPI3_HOST
   39 |                 &w5100_spi_handle ) );
      |                           ^~~~~~~~~~~~                                   
```

Lapshin, 2024-12-08:
@KaeLL , can you please simplify your example to only one file that can be compiled outside the IDF environment?

I tried to build `hello-world` example but replaced it with your code for `esp32` on commit https://github.com/espressif/esp-idf/commit/bcd80c92f3dd6c99cce7cc239afe8712bda04fac and did not get any error/warning message.

Do I understand correctly that you reported a bug with an incorrect line indication only?

igrr, 2024-12-08:
@Lapshin i guess you need to pull the latest changes to get https://github.com/espressif/esp-idf/commit/0d7e589a71268f7e177d0fa6892e70585b07df9e

KaeLL, 2024-12-09:
Be that as it may, still looks like an issue with GCC, no?

Lapshin, 2024-12-10:
@KaeLL , yes it's an issue that already improved for trunk (gcc's master branch) https://godbolt.org/z/Mr8bsoqa8

But it seems another issue appeared - double warnings output

Lapshin, 2024-12-10:
@KaeLL , you can track the issue here https://gcc.gnu.org/bugzilla/show_bug.cgi?id=117977

igrr, 2024-12-10:
True, but seems like that is not an ESP-specific issue, upstream GCC also doesn't place the marker correctly:
https://godbolt.org/z/31Ta18eMx

This upstream bug report seems related: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=102887.