espressif/crosstool-NG #72: libgcc does not support call0 ABI

Chalandi, 2025-03-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

libgcc.a must be provided for both ABI types (windowed, call0).

### Actual behavior (suspected bug)

the provided libgcc library in toolchain seems to be compiled only for windowed ABI, so far no support for call0 ABI.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

compile a C file that use float division with the compiler option "-mabi=call0"

### Project release version

14.2.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

PowerShell

Lapshin, 2025-03-04:
@Chalandi , yes, libgcc for esp chips is indeed compiled with only windowed ABI support. This is done due to esp-idf supports only the windowed ABI. Since migrating to CALL0 ABI requires to be supported in asm code including interrupt handling + ROM code that flashed to esp32 chips (which cannot be changed) - there is no CALL0 ABI support in the toolchain.

> compile a C file that use float division with the compiler option "-mabi=call0"

The toolchain can compile operations on floating-point numbers without the option "-mabi=call0". Why do you want to add this option? Could you please provide more details about what you want to achieve?

Chalandi, 2025-03-06:
Hi @Lapshin, I actually compiled the necessary libraries for the call0 ABI myself before opening the ticket.
That said, I really think the toolchain should provide them by default. Without these, using the toolchain with "-mabi=call0" leads to a non-working binary, no matter if the SDK is designed around the windowed ABI or not.

Lapshin, 2025-03-06:
@Chalandi , I can understand your point about expectations the toolchain should support all declared options.

But this toolchain is configured and distributed for Espressif customers that develop their firmware based on esp-idf sources (including arduino, platformio, ...). From this perspective, maintaining this ABI results in significant disk space consumption while providing little to no benefit to users.

I assume this will not be enabled until ESP-IDF supports it (which seems unlikely).