espressif/crosstool-NG #74: macOS Sequoia 15.5: Toolchain 15.1.0_20250607 is not working

pkarashchenko, 2025-06-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The C-code for ESP32S3 should compile.

### Actual behavior (suspected bug)

Getting error during compilation

### Error logs or terminal output

```shell
cc1: fatal error: 'XTENSA_GNU_CONFIG' is defined but could not be loaded: dlopen(/usr/local/lib/xtensa_esp32s3.so, 0x0001): tried: '/usr/local/lib/xtensa_esp32s3.so' (code signature in <B09EF242-055A-39DC-9949-B09A16D1D76C> '/usr/local/Cellar/x86_64-xtensa-esp-elf-gcc/15.1.0_20250607/gcc/lib/xtensa_esp32s3.so' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/xtensa_esp32s3.so' (no such file), '/usr/local/lib/xtensa_esp32s3.so' (code signature in <B09EF242-055A-39DC-9949-B09A16D1D76C> '/usr/local/Cellar/x86_64-xtensa-esp-elf-gcc/15.1.0_20250607/gcc/lib/xtensa_esp32s3.so' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/usr/local/Cellar/x86_64-xtensa-esp-elf-gcc/15.1.0_20250607/gcc/lib/xtensa_esp32s3.so' (code signature in <B09EF242-055A-39DC-9949-B09A16D1D76C> '/usr/local/Cellar/x86_64-xtensa-esp-elf-gcc/15.1.0_20250607/gcc/lib/xtensa_esp32s3.so' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/Cellar/x86_64-xtensa-esp-elf-gcc/15.1.0_20250607/gcc/lib/xtensa_esp32s3.so' (no such file), '/usr/local/Cellar/x86_64-xtensa-esp-elf-gcc/15.1.0_20250607/gcc/lib/xtensa_esp32s3.so' (code signature in <B09EF242-055A-39DC-9949-B09A16D1D76C> '/usr/local/Cellar/x86_64-xtensa-esp-elf-gcc/15.1.0_20250607/gcc/lib/xtensa_esp32s3.so' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs)
```

### Steps to reproduce the behavior

1. Write simple C function
2. Try to compile it with 15.1.0_20250607 using ESP32S3 target

### Project release version

15.1.0_20250607

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

Sequoia 15.5

### Shell

Bash

Lapshin, 2025-06-29:
@pkarashchenko , thanks for the report,

how did you install the toolchain? (Downloaded via browser or installed with install.sh script?)
Does previous toolchain work correctly on the current system?

Lapshin, 2025-06-29:
@pkarashchenko , I meant 14.2.0. It likely will the same behavior

Lapshin, 2025-06-29:
@pkarashchenko , could you try to install it using IDF install script and check how it works?

Lapshin, 2025-06-29:
@pkarashchenko, sorry for making a false promise. After reviewing the binaries, I couldn't find any issues on our side.

It seems something is wrong with your "brewery". I attempted to execute our releases, both ARM64 and x86_64 distributions (under `rosetta` emulation). And they work just fine.
Note: I tested on OsX 14.4 (don't want to upgrade).

Also, there is code-sign output info for files (`xtensa-esp-elf-as` loads `xtensa_esp32s3.so` library). And as you can see `TeamIdentifier=QWXF6GB4AV` is the same for both of them:

```
$ codesign -d --verbose=4 ./xtensa-esp-elf-as
Executable=/Users/xxx/Downloads/xtensa-esp-elf/bin/xtensa-esp-elf-as
Identifier=xtensa-esp-elf-as
Format=Mach-O thin (x86_64)
CodeDirectory v=20500 size=13021 flags=0x10000(runtime) hashes=401+2 location=embedded
VersionPlatform=1
VersionMin=658432
VersionSDK=786432
Hash type=sha256 size=32
CandidateCDHash sha256=77bf8c4d4be53ff77bd258e9bf7d0f6bd4a02f69
CandidateCDHashFull sha256=77bf8c4d4be53ff77bd258e9bf7d0f6bd4a02f69aa1b1595456f0c08a437da91
Hash choices=sha256
CMSDigest=77bf8c4d4be53ff77bd258e9bf7d0f6bd4a02f69aa1b1595456f0c08a437da91
CMSDigestType=2
Executable Segment base=0
Executable Segment limit=1409024
Executable Segment flags=0x1
Page size=4096
CDHash=77bf8c4d4be53ff77bd258e9bf7d0f6bd4a02f69
Signature size=9026
Authority=Developer ID Application: ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD. (QWXF6GB4AV)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=7 Jun 2025 at 23:04:22
Info.plist=not bound
TeamIdentifier=QWXF6GB4AV
Runtime Version=12.0.0
Sealed Resources=none
Internal requirements count=1 size=180
```
```
$ codesign -d --verbose=4 ../lib/xtensa_esp32s3.so
Executable=/Users/xxx/Downloads/xtensa-esp-elf/lib/xtensa_esp32s3.so
Identifier=xtensa_esp32s3
Format=Mach-O thin (x86_64)
CodeDirectory v=20500 size=5050 flags=0x10000(runtime) hashes=152+2 location=embedded
VersionPlatform=1
VersionMin=658432
VersionSDK=786432
Hash type=sha256 size=32
CandidateCDHash sha256=ad6c58360cceb18ae5ab566d9af5a42564d37e31
CandidateCDHashFull sha256=ad6c58360cceb18ae5ab566d9af5a42564d37e3131d72a6c567dd2b6fcb8093b
Hash choices=sha256
CMSDigest=ad6c58360cceb18ae5ab566d9af5a42564d37e3131d72a6c567dd2b6fcb8093b
CMSDigestType=2
Executable Segment base=0
Executable Segment limit=155648
Executable Segment flags=0x0
Page size=4096
CDHash=ad6c58360cceb18ae5ab566d9af5a42564d37e31
Signature size=9025
Authority=Developer ID Application: ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD. (QWXF6GB4AV)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=7 Jun 2025 at 23:04:37
Info.plist=not bound
TeamIdentifier=QWXF6GB4AV
Runtime Version=12.0.0
Sealed Resources=none
Internal requirements count=1 size=176
```

Please, try execute it without any modification:
 - download the toolchain from page https://github.com/espressif/crosstool-NG/releases/tag/esp-15.1.0_20250607
 - unpack
 - run gcc that you unpacked
 - please write what the result was

pkarashchenko, 2025-06-29:
I did a bit of research and found that homebrew patches shared libraries during installation step to ensure correct relocation :( But with signed libraries that obviously leads to getting invalid signature after patching. With manual download things work as expected. I will need to rework my way of managing the tools on macOS :) till now homebrew was very convenient, but seems not usable anymore. Let's close this issue

closed 2025-06-29 by pkarashchenko