espressif/crosstool-NG #75: Building C++ application fails to compile '__gthread_yield'

pkarashchenko, 2025-07-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The C++ application compiles with no issues.

### Actual behavior (suspected bug)

Building C++ application fails to compile '__gthread_yield'

### Error logs or terminal output

```shell
CXX:  test_project/test.cc In file included from /opt/gcc/xtensa-esp-elf/xtensa-esp-elf/include/c++/15.1.0/xtensa-esp-elf/esp32s3/no-rtti/bits/gthr.h:157,
                 from /opt/gcc/xtensa-esp-elf/xtensa-esp-elf/include/c++/15.1.0/bits/std_mutex.h:43,
                 from /opt/gcc/xtensa-esp-elf/xtensa-esp-elf/include/c++/15.1.0/condition_variable:44,
                 from /workspace/test_project/test.h:22,
                 from test_project/test.cc:18:
/opt/gcc/xtensa-esp-elf/xtensa-esp-elf/include/c++/15.1.0/xtensa-esp-elf/esp32s3/no-rtti/bits/gthr-default.h: In function 'int __gthread_yield()':
/opt/gcc/xtensa-esp-elf/xtensa-esp-elf/include/c++/15.1.0/xtensa-esp-elf/esp32s3/no-rtti/bits/gthr-default.h:739:10: error: 'sched_yield' was not declared in this scope; did you mean 'pthread_yield'?
  739 |   return __gthrw_(sched_yield) ();
```

### Steps to reproduce the behavior

I'm developing C++ application for NuttX OS. It has `sched.h` available. My application uses `std::mutex`.
I've inspected the content of `xtensa-esp-elf/include/c++/15.1.0/xtensa-esp-elf/esp32s3/no-rtti/bits/gthr-default.h` and see that it has lines
```
#ifdef HAVE_SCHED_H
# include <sched.h>
#endif
```
But adding `-D HAVE_SCHED_H` to compilation flags somehow does not help. For now I worked around the failure by adding explicit `#include <sched.h>` in prior to including `#include <mutex>`, but believe that this issue should be fixed on toolchain side. Not sure, but maybe switching to `pthread_yield` may help here.

### Project release version

15.1.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

macOS Sequoia 15.5

### Shell

ZSH

igrr, 2025-07-14:
We can probably add this header on IDF side, seems like most of the functions can be simple stubs.

It would be convenient to be able to continue using the same toolchain builds for IDF and for NuttX.

Lapshin, 2025-07-30:
@igrr , that could be done, but it might be only one of many cases. Coordination with the NuttX team is necessary.