espressif/crosstool-NG #78: gcc 15.2 fails to compile with IDF 5.5 release branch

Jason2866, 2025-09-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

To compile succesfully the DSP component

### Actual behavior (suspected bug)

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/managed_components/espressif__esp-dsp/modules/matrix/mul/fixed/dspm_mult_s16_arp4.S:99: Error: illegal operands `esp.srcmb.s16.qacc q2,x26,0'

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

update IDF tools.json with gcc15.2 and compile a project with DSP

### Project release version

v15.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Shell

ZSH

### Additional context

CI run of fork from Arduino Lib Builder https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/17894582722/job/50879500066

Lapshin, 2025-09-21:
@Jason2866 ,

1. If you use toolchain releases that are different from what's declared in tools.json, it's entirely your responsibility.
2. Release GCC 15.2.0 uses espv spec 2p2 by default for the next chip revision (chip is not released yet). To use the instructions set for old chips you may perform 2 options:
  - add `-mespv-spec=2p1` to compile options
  - use `xespv2p1` explicitly in `-march` compile option

I strongly recommend you not use gcc 15 in IDF versions less than v6.0. See more details here: https://github.com/espressif/crosstool-NG/issues/76#issuecomment-3139937498

Jason2866, 2025-09-21:
Thx for the solution. This `add -mespv-spec=2p1 to compile options` solves.